pimoroni/displayotron #10: Wrong text after run demo script

remik, 2015-02-18:
If I ran script `sudo python hello_world.py` then I see only "ello World". 
If I ran this manually in console line by line then I see "Hello World".

I have new Pi2 - is my device too fast? :P

Gadgetoid, 2015-02-18:
Haha, that's a good hypothesis! I've seen this problem too, but only briefly enough to brush it off as part of my brutal testing on Pi 2. I'll have to look into it in more detail.

Gadgetoid, 2015-02-19:
I've updated ST7036, the LCD driver library, update it with:

    sudo pip install st7036 --ugprade

And see if your problem goes away!

remik, 2015-02-20:
You have typo in command, should be:
`sudo pip install st7036 --upgrade`

Yes, it is working. Thanks!

closed 2015-02-20 by remik