pimoroni/displayotron #11: BananaPro support

minorbug, 2015-04-04:
The Banana Pi and Banana Pro (which I have) have recently added GPIO support through its own library (https://github.com/LeMaker/RPi.GPIO_BP). I've been trying to get my dot3k to work with the Banana Pro without much luck.

Gadgetoid, 2015-04-27:
I unfortunately don't have a Banana Pi or Pro to test dot3k with, and it's not high on my list of priorities, but if I can help in any way ( I don't know much about the Nana so probably not :( ) I will.

minorbug, 2015-09-09:
For those interested, I was able to get the DOT3K to work by making a few changes to sn3218.py and /etc/modules. I've documented the instructions here: http://mattbaker.me/2015/09/08/displayotron-3000-for-banana-pro/

closed 2016-02-24 by Gadgetoid