pimoroni/displayotron #15: Constant CPU usage of at least 20%

jorisvervuurt, 2015-10-24:
When using any of the `dothat.touch` event handlers (e.g. `@touch.on(touch.UP)`), while running `top` I noticed that the `python` process is constantly using at least 20% CPU. This is the case even when I don't touch any of the buttons on the Display-O-Tron HAT. When I removed the `@touch.on` event handlers, CPU usage went down to pretty much 0%.

I tested the same thing with the `dot3k.joystick` event handlers (e.g. `@joystick.on(joystick.UP)`), but those seem to work just fine. Pretty much no CPU load while idle, so it looks like this problem only exists for the touch event handlers.

Could you guys please take a look at this?
Thank you very much in advance!

Gadgetoid, 2015-10-25:
The area of interest for this problem is the interrupt polling loop here: https://github.com/pimoroni/cap1xxx/blob/master/library/cap1xxx.py#L360-L371

I suspect that delay needs increasing- in fact there must be a minimum sensible time for it which I could look up or derive. I kept it low because I wanted a good response in Piano HAT, but never tried raising it.

Piano HAT uses the alert pins, however, so the GPIO read is way less expensive than the round trip to an i2c register that the library needs to make for DotHAT.

Gadgetoid, 2015-10-27:
I've updated cap1xxx to increase the delay from a microsecond ( completely bonkers and probably not even meaningful in Python ) to 5ms, which I think should be plenty fast enough for even Piano HAT. https://github.com/pimoroni/cap1xxx/commit/ef34015813302e05dd11a5f7b440a3cff42e4530

If you grab the cap1xxx library from GitHub and install it, hopefully your CPU usage will drop dramatically.

jorisvervuurt, 2015-10-30:
Cool.
Will re-running the dot3k installation script update the cap1xxx library?

I'll take a look at it ASAP. ;-)

Gadgetoid, 2015-10-30:
Nope, you'd have to git clone the cap1xxx library and install that separately.

jorisvervuurt, 2015-10-30:
Ah, okay.
Maybe a stupid question, but where and how do I install it?

Also, since other Display-O-Tron HAT might not be aware of the CPU usage, you might want to make sure that re-running the dot3k installation script also updates the cap1xxx library. ;-)

Gadgetoid, 2015-10-30:
Under normal circumstances it should go, but I haven't pushed the change up to pip yet. Try something like:

```bash
git clone https://github.com/pimoroni/cap1xxx
cd cap1xxx/library
sudo python setup.py install
sudo python3 setup.py install
```

capfuji, 2016-02-03:
I was wondering, is this issue solved in the official installation? I still have CPU usage around 20% with active dothat.touch event handlers, but it could also be my crappy Python code :sweat_smile:

wujku, 2016-02-03:
Issue still exists.

Gadgetoid, 2016-02-24:
I've pushed up version 0.1.1 to pip which should contain this fix: https://pypi.python.org/pypi/Cap1xxx

Would be interested in feedback! @capfuji @wujku

capfuji, 2016-02-24:
Thanks! CPU usage is down from around 20 to 2.3 %. Touch buttons are still working as expected. Perfect 😊

jorisvervuurt, 2016-03-06:
I guess this issue can be closed. Good work!

closed 2016-03-06 by jorisvervuurt