pimoroni/displayotron #16: I think your current folder structure is different when run from your install command

contractorwolf, 2015-11-25:
I see on this page that you say to run this command to install the Pimoroni on the RPi:
\curl -sS get.pimoroni.com/dot3k | bash

but the folder structure that is described in this repo is not what I get on the RPi, as seen in this picture here:
https://lh3.googleusercontent.com/A1FBFdrD7IlMqXfJR-25DAe3tGXKmGDtastnbvkZ8odA9r8-NIsz6iRbLQs6gcD9wYnVX9JzFBV40958G1ufmYcc43kwJT88-b4xA-pGlnd-MdM80fYHFSL4TxdkNcm84XsFsQPy7gqnFMewnzmSAbONdF6Rrq8e6uo3PyToFzs1jGCZ9YgS1-GQJLs9GdvNPh-sRKe8wNKPDn4u6E3RkD74aeqwy2OFPXqzPfls5Zj-CR2iWJ1gcfzGE34mi4WPiIjCnFnqibutmYU26S-fqUyTNrVaylY0I7VxrBmkHYfxv2_4b5DfeP0fs-RwM_qutoKTFmQ0V6Zj3BOkRbvwGRfuZYZQRDcAn3Up0zJwfQR2v7A2QTYS1_zGHjvX_0fRu8bpavHV95N0WU05K-8vEVsqbUANlb4f5zecXyw18rZGErvtS6BJgR7H7EgiSGYJF3QclTEoZJzCIPidPMO4EwT_zPHX1gXzxcY-30j-7DksXg5VcxmNlqnxpM_V03ur-J9_qdJ0-Ewhelphgvk9OVswsCUCsmfziFj_99SkJal3Is2hNx1x-0Pdiq_r=w1526-h935-no

I was finally able to figure it out, but you should fix the documentation or the installer so they are showing the same
Thanks!

RogueM, 2015-11-26:
Your link does not work for me, it seems to be restricted/private.

contractorwolf, 2015-11-29:
I attached it to the email, let me know if you can see itthanksJames

James Wolf 
http://contractorwolf.com 

"Worker bees can leave 
Even drones can fly away 
The queen is their slave" 

Date: Wed, 25 Nov 2015 23:56:57 -0800

Gadgetoid, 2015-11-30:
Unfortunately we can't see it- replying to the email has updated the issue on github which doesn't support attachments.

contractorwolf, 2015-11-30:
try the link above again, I can see it no prob in incognito mode.  Anyways the folder structure that I see looks like this:
Pimoroni/dot3k/dot3k/basic

RogueM, 2015-11-30:
yes, the examples are copied in a subfolder named 'dot3k' currently. One of it's subfolder is also named 'dot3k', as per original display-o-tron.

It will change soon and the roit folder for the examples will be renamed to avoid tye confusion. But I guess the main point is that the Pimoroni/dot3k folder in your home directory is not a copy of this repo, and is not meant to be... does this answer your query?

RogueM, 2015-11-30:
btw, you do not need nor really want to clone this repo to your computer, unless you intend to edit the library for your own needs or contribute fixes, the curl/bash 'one-line installer is provided for the very purpose of copying all the components in the right place on your system, as well as fulfilling dependencies they might require to operate.

closed 2015-12-25 by contractorwolf