pimoroni/displayotron #18: i dont get it, can you explain the list of 8 bit integers?

Floppynator, 2016-01-06:
Can you explain where i can get or generate this?

Gadgetoid, 2016-01-06:
Are you talking about the integers used for animations?

Floppynator, 2016-01-07:
Yes sir. Can you explain this please?

Floppynator, 2016-01-20:
no one? how did you make the pacman?

pcarver, 2016-02-04:
First get yourself some graph paper, or open a spreadsheet program and resize the columns and rows to the same size so you have a grid of squares. Notice that on the LCD there are three rows of rectangles, look more closely and notice that each rectangle is made up eight rows of squares, five across.

On your graph paper, draw out your pictures using eight rows of five squares across for each picture. Each square can either be black or white, on or off.

The easiest way to represent this in python is to convert eight character strings to integers using base two. Create strings as a sequence of eight ones and zeros using one to represent the black squares and zero to represent the white squares. Convert the strings to integers using the int function, with a parameter to indicate that the strings are base two (i.e. binary, ones and zeros). Arrange the strings into a list with each string of eight bits representing one row in one rectangle on the LCD. The left three bits in each string don't matter because the LCD rectangles are only five squares wide so they use the rightmost five bits.

The pacman image is reprsented in condensed form using hexadecimal (base sixteen), but base two (binary) is easier to visualize. Here's an example:

char_map_happy = [ int('00000000', 2),
                   int('00000000', 2),
                   int('00001010', 2),
                   int('00000000', 2),
                   int('00010001', 2),
                   int('00001110', 2),
                   int('00000000', 2),
                   int('00000000', 2)]

And for an animation you would use a list of lists, so each frame is a list of eight strings of eight bits.

anim      = [ int('00000000', 2),
                   int('00000000', 2),
                   int('00001010', 2),
                   int('00000000', 2),
                   int('00010001', 2),
                   int('00001110', 2),
                   int('00000000', 2),
                   int('00000000', 2)],

                 [ int('00000000', 2),
                   int('00000000', 2),
                   int('00001010', 2),
                   int('00000000', 2),
                   int('00001110', 2),
                   int('00011111', 2),
                   int('00001110', 2),
                   int('00000000', 2)]
                   ]

Floppynator, 2016-02-04:
wow. Thank you for this detailed answer! 

cheers chris

AndiCui, 2016-08-30:
This is 7 month old. @gadetoid make a document from the answer and close this?

pcarver, 2016-11-14:
I made some revisions to this write-up and put it into a file.

https://github.com/pimoroni/dot3k/pull/42

AndiCui, 2016-11-14:
@pcarver Nice work!
One tiny thing for @Gadgetoid to think though
I prefer this style of binary number, since it is a bit less clunky.
```python
[0b00000,
# ...
]
```
@Gadgetoid What do you think the style in doc should be?

AndiCui, 2016-11-17:
Merged #42

closed 2017-11-20 by Gadgetoid