pimoroni/displayotron #23: support for Python 3.5

Kowlin, 2016-05-24:
Will the 3k and hat ever support Python 3.5? I'm fully aware that Raspian doesn't officially support Python 3.5. But I was wondering if there are any plans for a 3.5 version due the fact that people (my self included) can easily build Python3.5 for them self's.

Gadgetoid, 2016-06-08:
It's definitely not on my list of priorities at the moment, since the vast majority of people will be running whatever the default is and time is precious. But if you've got any idea where the incompatibilities lie and know how to fix them in a backwards-compatible way ( I'd assume at least some knowledge from anyone compiling Python 3.5 from source :D ) then I'd welcome a PR!

Kowlin, 2016-06-08:
Its been some times since I seen the error. But I'll take a look on where the error will originate from.

Gadgetoid, 2017-02-07:
It could have been an issue with the sn3218 library, although I'm not sure it would happen during non test use: https://github.com/pimoroni/sn3218/pull/9

I think there are a few Python 3.5 incompatibilities lurking still.

closed 2017-11-20 by Gadgetoid