pimoroni/displayotron #26: Alignment Issues

Nixpower, 2016-06-21:
Hi, I'm using a Displayotron 3000 on a Raspberry Pi 3 and I'm having some difficulty having the text characters align properly, as when I run the sample "Hello World" script, the display cuts off some characters and displays "ello World". Changing the script to print " Hello World" makes it work. However, the alignment issues are different for different sets of characters, as some will only be off by 1 character while others will be offset by 2 or 3.

Thanks,
Nixpower

Nixpower, 2016-06-22:
It looks like the issue I'm having is only the case when you don't use `backlight.rgb` prior to printing your text to the display. This looks like a really interesting and specific bug...

Gadgetoid, 2016-07-07:
I've seen this happen a couple of times on Display-o-Tron 3000, but haven't been able to narrow down exactly why. It's always been one character offset for me, and can generally be fixed with a call to ( if I remember correctly ) `home`

AndiCui, 2016-08-25:
I notice the same problem when I am clearing the display directly before other commands (usually set_cursor). A 10ms sleep solves the problem.

closed 2017-03-17 by Gadgetoid