pimoroni/displayotron #32: [Suggestion] A class for "tables"

AndiCui, 2016-08-27:
There are some use cases (like transit time table, etc), where it is useful to devide the display in to columns. (For example: Line Num, Destination, Remianing Mins)
It would be really cool if this can be done with the flexibility to set L/R-Aligned, and a scroll method for longer stuff (In this case destination)

AndiCui, 2016-08-27:
I did a quick “illustration”, so I can get the point across.
It DOES look great. (At least for me)
[image]

PS. Happy long weekend for those are in the UK

Gadgetoid, 2016-08-27:
That's quite tidy, and potentially useful. It might be worth building as a separate library with an accompanying example, rather than adding too much complexity to the base library though.

Features like auto-scrolling the text would be cool, too.

For UX there's a lot you can do with the 8 custom icons and tidy text placement, too. Plus the character table embedded in the display has boxes, chevrons and other useful tidbits: http://www.lcd-module.de/pdf/doma/dog-m.pdf

AndiCui, 2016-08-27:
Let's see when I have time to write my dodgy python code. :)
I will paste a link here if I manage to start it.
Cheers

AndiCui, 2016-08-27:
https://github.com/AndiCui/DOT-Table

AndiCui, 2016-08-27:
The lib is 80% done.
Still need some clean-up and testing with self defined character.
Consider transfer owner to the official account?

Gadgetoid, 2017-11-20:
This is good stuff- is there any way we can bring more attention to this in the Display-o-Tron repo?

closed 2018-02-07 by Gadgetoid