pimoroni/displayotron #35: PermissionError when using without root/sudo

AxelRHD, 2016-10-05:
Always getting an PermissionError when trying to import the lcd-module:

Tried to fix it with file-permissions in the lib-directory, but that does not seem to be the problem.

Here is the full error:

`Traceback (most recent call last):
  File "touch.py", line 10, in <module>
    import dothat.lcd as l
  File "/usr/local/lib/python3.4/dist-packages/dothat/lcd.py", line 6, in <module>
    lcd = st7036.st7036(register_select_pin=25, reset_pin=12)
  File "/usr/local/lib/python3.4/dist-packages/st7036.py", line 40, in __init__
    self.spi.open(0, spi_chip_select)
PermissionError: [Errno 13] Permission denied`

AxelRHD, 2016-10-05:
"Fixed" it by myself...

I reinstalled the OS for other reasons with

`curl -sS get.pimoroni.com/displayotron | bash`

The first time I think I accidentally used

`curl -sS get.pimoroni.com/displayotron | sudo bash`

All fine now.

Gadgetoid, 2016-10-13:
That's curious. I can't see how that would make a difference. @RogueM might know!

RogueM, 2016-10-13:
Can't think of why no, besides copying the examples from the repo that probably shouldn't be done with elevated privileges, the rest is and should be installed with sudo.

Can you access os-installed modules from within a venv though? If you ran the script while in a venv I suspect you'd end up with something not quite right, I've never tried it.

closed 2017-03-17 by Gadgetoid