pimoroni/displayotron #46: How can I reference an individual LED on dothat?

juniorRubyist, 2016-12-28:
Can I reference an individual backlight LED in the dothat library so I can get something like this? The rows are the individual lights, (R = Red, B = Blue)
```
[ 1 | 2 | 3 | 4 | 5 | 6 ]
[ R | B | R | B | R | B ]
```

RogueM, 2016-12-29:
you can address the backlight as leds as follows:

`set(index, value):

    Set a specific LED to a value
    
    Args:
        index (int): index of the LED from 0 to 18
        value (int): brightness value from 0 to 255`

they run as BGR zone 1, 0-2, BGR zone 2, 3-5, etc

so something like this would get zone 1-2 as RB

```python
import dothat.backlight as backlight

backlight.set(2,128)
backlight.set(3,128)
```

juniorRubyist, 2016-12-30:
Sorry, I was a bit unclear. Is it possible to do this with RGB?

juniorRubyist, 2016-12-30:
@RogueM I found a `single_rgb` function in [`dot3k/library/dothat/single_rgb`, Line 226](https://github.com/pimoroni/dot3k/blob/master/library/dothat/backlight.py#L226). I can use it like so to solve my problem:
```python
backlight.single_rgb(0, 209, 85, 66)
backlight.single_rgb(1, 66, 109, 209)
backlight.single_rgb(2, 209, 85, 66)
backlight.single_rgb(3, 66, 109, 209)
backlight.single_rgb(4, 209, 85, 66)
backlight.single_rgb(5, 66, 109, 209)
```

They should really document this.

closed 2016-12-30 by juniorRubyist