pimoroni/displayotron #51: Install one-liner not fetching up-to-date examples

ncguk, 2017-08-07:
I've just installed the Display-o-Tron software using the `curl -sS get.pimoroni.com/displayotron | bash` one-liner (on a Pi Zero W, Raspbian Jessie, with everything up to date), and have noticed that there are fewer examples in the resulting 'full install' than there are in the main branch of the GitHub repo, specifically in the `basic` folder of both the `dot3k` and `dothat` folders.

Changing `copyhead="no"` to `copyhead="yes"` in the installer bash script solves this, but I'm guessing there is a better solution than that.

The same is also true for the Blinkt library. I'll file a separate issue for that.

Gadgetoid, 2017-08-17:
This is by design, rather than by accident, since there may be cases when the latest examples in GitHub rely upon features that aren't available in the shipped versions of the library.

There may be a better way to manage this, but for the time being the installer will usually only pull examples that were available at the time the latest release was tagged.

ncguk, 2017-08-19:
Okay, no worries. I'll close the issue.

closed 2017-08-19 by ncguk