pimoroni/displayotron #52: Experimental support warning with Raspbian Stretch

ncguk, 2017-08-20:
Installing the Display-o-Tron software using the one-liner on the new Raspbian Stretch image generates the "Support for your operating system is experimental. Please visit forums.pimoroni.com if you experience issues with this product" warning (installation continues). I imagine this extends to all the Pimoroni one-line installers. Is this something that should be fixed, or is Stretch testing not complete?

Also, in this instance, the installer complains that it's "Unable to install smbus for python 3!"

Gadgetoid, 2017-08-21:
Support is, indeed, experimental- as you've found out!

The "Unable to install smbus for python 3!" and possibly even "Unable to install smbus for python 2!" messages may be incorrect- a quick look into the guts of the installer suggest it's using `pip list | grep smbus` to validate that the package exists. However both this, and freeze, appear to list only packages installed and managed by pip, and not system-owned packages.

I get the same messages, but if I ask apt if they installed:

```
pi@raspberrypi:~ $ apt list --installed | grep smbus

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

python-smbus/stable,now 3.1.2-3 armhf [installed]
python3-smbus/stable,now 3.1.2-3 armhf [installed]
```

ncguk, 2017-08-21:
Yeah. I used `pip3 list` and saw smbus wasn't installed. I imagine this will baffle/alarm some new users. Is there an ETA for Stretch support?

Gadgetoid, 2017-12-18:
I've finally found the time to track down the cause of:

> "Unable to install smbus for python 3!"

It looks like version 3.1.2 of `python-smbus` quietly dropped the `.egg-info` file which is required for `pip` to "see" a system managed Python package.

Due to the way our installer checks for the existence of this package, after trying various methods to install it, this change (it appears to be a bug since it's not mentioned in the changelog) breaks our scripts :(

I've reported the bug- assuming it is legitimately a bug -but in the meantime I'll have to revise our scripts to work around it.

closed 2018-03-01 by Gadgetoid