pimoroni/displayotron #58: Displayotron HAT No Response - Blank Screen - Under many circumstances

fc-gituser, 2018-12-15:
Dear Forum, Relative to this Post (which I do not completely understand what is to be done) please respond to my Displayotron Problems

I have two Displayotron HATS. I had one working at one time. Currently, neither of them are responsive after installing three different ways; through Pimoroni Dashboard, the one-liner starting with curl -sS, and the one-liner starting with curl get.pimoroni.***. Both Raspberry Pi Zero W or Raspberry Pi 2 were tested. Each have been attempts with either Stretch or Jessie. 

On all these attempts, the Raspberry Pi operates normally with several different programs functioning successfully. Under all these attempts, the device has no output and screen is blank. I must be missing something.  

The following is a session output to explain the problem: 
Please help, Thank You, Frank. 

pi@raspberrypi:~ $ ls
adafruit-pitft.sh  Downloads                          Pictures  python_games
Desktop            install-piface-real-time-clock.sh  Pimoroni  Templates
Documents          Music                              Public    Videos
pi@raspberrypi:~ $ cd Pimoroni
pi@raspberrypi:~/Pimoroni $ ls
displayotron
pi@raspberrypi:~/Pimoroni $ cd displayotron/examples/dothat/basic
pi@raspberrypi:~/Pimoroni/displayotron/examples/dothat/basic $ ls
backlight.py  graph.py  hello_world.py  ipaddr.py  room.py  touch.py
pi@raspberrypi:~/Pimoroni/displayotron/examples/dothat/basic $ python touch.py
Traceback (most recent call last):
  File "touch.py", line 6, in <module>
    import dothat.lcd as lcd
  File "/usr/local/lib/python2.7/dist-packages/dothat/lcd.py", line 6, in <module>
    lcd = st7036.st7036(register_select_pin=25, reset_pin=12)
  File "/usr/lib/python2.7/dist-packages/st7036.py", line 42, in __init__
    self.spi.open(0, spi_chip_select)
IOError: [Errno 2] No such file or directory
pi@raspberrypi:~/Pimoroni/displayotron/examples/dothat/basic $ python hello_world.py
Traceback (most recent call last):
  File "hello_world.py", line 3, in <module>
    import dothat.lcd as lcd
  File "/usr/local/lib/python2.7/dist-packages/dothat/lcd.py", line 6, in <module>
    lcd = st7036.st7036(register_select_pin=25, reset_pin=12)
  File "/usr/lib/python2.7/dist-packages/st7036.py", line 42, in __init__
    self.spi.open(0, spi_chip_select)
IOError: [Errno 2] No such file or directory
pi@raspberrypi:~/Pimoroni/displayotron/examples/dothat/basic $ python backlight.py
Traceback (most recent call last):
  File "backlight.py", line 7, in <module>
    import dothat.lcd as lcd
  File "/usr/local/lib/python2.7/dist-packages/dothat/lcd.py", line 6, in <module>
    lcd = st7036.st7036(register_select_pin=25, reset_pin=12)
  File "/usr/lib/python2.7/dist-packages/st7036.py", line 42, in __init__
    self.spi.open(0, spi_chip_select)
IOError: [Errno 2] No such file or directory
pi@raspberrypi:~/Pimoroni/displayotron/examples/dothat/basic $

fc-gituser, 2018-12-16:
The above issue posted by myself was in response to another post, "Experimental support warning with Raspbian Stretch  #52" which had been closed. Itcan be reached directly at: https://github.com/pimoroni/displayotron/issues/52
I didn't really understand that post but thought it might explain the reason I have been having Displayotron Hat problems. For more experienced users, please interpret that post relative to the above issues and direct me as to what I may doing wrong. Thankk you, Frank.

Gadgetoid, 2020-03-10:
Sorry this issue had slipped my notice. I hope you've managed to get it resolved via support or otherwise since, but please feel free to re-open this if not.

closed 2020-03-10 by Gadgetoid