pimoroni/displayotron #63: Can we use this library on bullseye?

mmozzano, 2021-11-11:
Trying to install the library on a fresh install of bullseye by running:

curl -sS get.pimoroni.com/displayotron | bash

I get the following error:

```
Selecting previously unselected package python3-spidev.
(Reading database ... 45053 files and directories currently installed.)
Preparing to unpack .../python3-spidev_2.0~git20150907_armhf.deb ...
Unpacking python3-spidev (2.0~git20150907) ...
dpkg: dependency problems prevent configuration of python3-spidev:
 python3-spidev depends on python3 (<< 3.5); however:
  Version of python3 on system is 3.9.2-3.
```

Looks like bullseye comes with Python 3.9 but the DisplayoTron library requires 3.5 or lower?

Gadgetoid, 2021-11-11:
I think the bash installer probably tries to install a downloaded .deb file, rather than rely on the system spidev.

You should just be able to:

```
sudo apt install python3-spidev
python3 -m pip install dot3k
```

And then grab the examples from this repository.

mmozzano, 2021-11-11:
Thanks for the response.  I've tried that and get the following error:

```
mmm@pi-4-8gb:~ $ sudo apt install python3-spidev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python3-spidev : Depends: python3 (< 3.8) but 3.9.2-3 is to be installed
E: Unable to correct problems, you have held broken packages.

```

ahnlak, 2021-11-11:
Looks like they've bumped the Python3 version in Bullseye but haven't updated the `spidev` requirements to match. I guess that's an rpi.org job?

mmozzano, 2021-11-11:
I have yes...
```

mmm@pi-4-8gb:~ $ sudo apt update
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:3 http://packages.azlux.fr/debian buster InRelease
Get:4 https://download.docker.com/linux/raspbian buster InRelease [33.6 kB]
Fetched 33.6 kB in 1s (23.2 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
mmm@pi-4-8gb:~ $ sudo apt install python3-spidev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python3-spidev : Depends: python3 (< 3.8) but 3.9.2-3 is to be installed
E: Unable to correct problems, you have held broken packages.

```

Is my list of repos pulling the list from buster first? 

mmm@pi-4-8gb:~ $ cat /etc/apt/sources.list
deb http://raspbian.raspberrypi.org/raspbian/ bullseye main contrib non-free rpi
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi

Gadgetoid, 2021-11-11:
> Is my list of repos pulling the list from buster first?

eeek! 😬  just do a clean SD card reinstall from the official image, Bullseye is broken enough without inviting more problems 😆

mmozzano, 2021-11-11:
On it now, will report back shortly.

ahnlak, 2021-11-11:
I would say yes, buster is (for some reason) overriding bullseye - the buster `spidev` requires a python << 3.8.

Upgrades across releases always seem to throw up problems like this :-(

mmozzano, 2021-11-11:
A fresh install of bullseye has resolved the issue!  I am able to run the installer and it ran through fine.

```
mmm@pi-4-8gb:~ $ curl -sS get.pimoroni.com/displayotron | bash

This script will install everything needed to use
Display-o-Tron 3000/HAT

Always be careful when running scripts and commands copied
from the internet. Ensure they are from a trusted source.

If you want to see what this script does before running it,
you should run: 'curl https://get.pimoroni.com/displayotron'

Note: Display-o-Tron 3000/HAT requires SPI communication
Note: Display-o-Tron 3000/HAT requires I2C communication

Do you wish to continue? [y/N] y

Checking environment...
Updating apt indexes...
.....................

Checking hardware requirements...

Checking for packages required for GPIO control...
raspi-gpio is already installed
RPi.GPIO installed and up-to-date

SPI must be enabled for Display-o-Tron 3000/HAT to work
SPI is now enabled

Checking packages required by SPI interface...
spidev installed and up-to-date

I2C already enabled

Checking packages required by I2C interface...
Python 2 smbus installed and up-to-date
Python 3 smbus installed and up-to-date

Display-o-Tron 3000/HAT comes with examples and documentation that you may wish to install.
Performing a full install will ensure those resources are installed,
along with all required dependencies. It may however take a while!

Do you wish to perform a full install? [y/N] n

Checking for dependencies...
python3-sn3218 is already installed
python3-st7036 is required
Installing python3-st7036...
E: Unable to locate package python3-st7036
Apt failed to install python3-st7036!
Falling back on pypi...

Installing Display-o-Tron 3000/HAT library for Python 3...

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting dot3k
  Downloading https://www.piwheels.org/simple/dot3k/dot3k-2.0.1-py3-none-any.whl (15 kB)
Requirement already satisfied: sn3218 in /usr/lib/python3/dist-packages (from dot3k) (1.2.7)
Collecting st7036
  Downloading https://www.piwheels.org/simple/st7036/st7036-1.4.4-py3-none-any.whl (5.6 kB)
Requirement already satisfied: cap1xxx in /usr/lib/python3/dist-packages (from dot3k) (0.1.3)
Requirement already satisfied: RPi.GPIO in /usr/lib/python3/dist-packages (from cap1xxx->dot3k) (0.7.0)
Requirement already satisfied: spidev in /usr/lib/python3/dist-packages (from st7036->dot3k) (3.5)
Installing collected packages: st7036, dot3k
Successfully installed dot3k-2.0.1 st7036-1.4.4

All done. Enjoy your Display-o-Tron 3000/HAT!

```

Thank you both, I'll close this issue.

closed 2021-11-11 by mmozzano