pimoroni/displayotron #64: AttributeError: 'AsyncWorker' object has no attribute 'isAlive'

mmozzano, 2021-11-11:
Fresh Bullseye install, runnig Python 3.9.2

mmozzano@pi-red-coupe:~/python_scripts $ python --version
Python 3.9.2

Install of DisplayOTron library completes successfully with no issues but when I try and run code I see following error:

```
Traceback (most recent call last):
  File "/home/mmozzano/python_scripts/touch.py", line 95, in <module>
    def handle_up(ch, evt):
  File "/usr/local/lib/python3.9/dist-packages/dothat/touch.py", line 42, in reg                                                                                                        ister
    _cap1166.on(channel=button, event='press', handler=handler)
  File "/usr/lib/python3/dist-packages/cap1xxx.py", line 386, in on
    self.start_watching()
  File "/usr/lib/python3/dist-packages/cap1xxx.py", line 400, in start_watching
    self.async_poll.start()
  File "/usr/lib/python3/dist-packages/cap1xxx.py", line 208, in start
    if self.isAlive() == False:
AttributeError: 'AsyncWorker' object has no attribute 'isAlive'
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/cap1xxx.py", line 409, in stop_watching
    self.async_poll.stop()
  File "/usr/lib/python3/dist-packages/cap1xxx.py", line 213, in stop
    if self.isAlive() == True:
AttributeError: 'AsyncWorker' object has no attribute 'isAlive'
Exception ignored in: <function Cap1xxx.__del__ at 0x76894a48>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/cap1xxx.py", line 522, in __del__
  File "/usr/lib/python3/dist-packages/cap1xxx.py", line 409, in stop_watching
  File "/usr/lib/python3/dist-packages/cap1xxx.py", line 213, in stop
AttributeError: 'AsyncWorker' object has no attribute 'isAlive'

```

This URL seems to imply the function isAlive has been changed to is_alive under Python 3.9 in file cap1xxx.py -

https://github.com/jupyter-vim/jupyter-vim/issues/51

Sure enough manually making this change to /usr/lib/python3/dist-packages/cap1xxx.py and the DisplayOTron works as usual.  Is this an issue with the DisplayOTron library itself?

Gadgetoid, 2021-11-11:
I could have sworn I made/released this fix in cap1xxx. See: https://pypi.org/project/Cap1xxx/

Can you run:

```
python3 -m pip freeze | grep cap1xxx
```

And let me know if you're on v0.1.4?

mmozzano, 2021-11-11:
It isnt listed as an installed library when I run the above.  

I do see your updated code change in the Cap1xxx library but that isnt reflected in what was pulled down when I ran the install script a few hours ago.

AmadeusW, 2022-01-07:
I ran into the same issue.
@mmozzano I was able to upgrade Cap1xxx only when explicitly specifying the version. Note that `sudo pip install Cap1xxx` didn't accomplish anything

```
pi@bench:~/Pimoroni/explorerhat/examples $ python --version
Python 3.9.2

pi@bench:~/Pimoroni/explorerhat/examples $ pip show Cap1xxx
Name: Cap1xxx
Version: 0.1.3
Summary: A module to drive various Microchip cap1xxx touch ICs
Home-page: http://shop.pimoroni.com
Author: Philip Howard
Author-email: phil@pimoroni.com
License: MIT
Location: /usr/lib/python3/dist-packages
Requires: RPi.GPIO
Required-by: ExplorerHAT

pi@bench:~/Pimoroni/explorerhat/examples $ sudo pip install Cap1xxx
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: Cap1xxx in /usr/lib/python3/dist-packages (0.1.3)
Requirement already satisfied: RPi.GPIO in /usr/lib/python3/dist-packages (from Cap1xxx) (0.7.0)

pi@bench:~/Pimoroni/explorerhat/examples $ sudo pip install Cap1xxx==0.1.4
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting Cap1xxx==0.1.4
  Downloading https://www.piwheels.org/simple/cap1xxx/Cap1xxx-0.1.4-py3-none-any.whl (8.5 kB)
Requirement already satisfied: RPi.GPIO in /usr/lib/python3/dist-packages (from Cap1xxx==0.1.4) (0.7.0)
Installing collected packages: Cap1xxx
  Attempting uninstall: Cap1xxx
    Found existing installation: Cap1xxx 0.1.3
    Not uninstalling cap1xxx at /usr/lib/python3/dist-packages, outside environment /usr
    Can't uninstall 'Cap1xxx'. No files were found to uninstall.
Successfully installed Cap1xxx-0.1.4
```

mmozzano, 2022-01-07:
Looks like the command I was running to get the version of Cap1xxx was wrong - I was using lower case c.

_<name>@pi-red-coupe:~ $ python3 -m pip freeze | grep Cap1xxx
Cap1xxx==0.1.4_

So I now have 0.1.4.

I uninstalled and then reinstalled the package by:

_pip install Cap1xxx==0.1.4_

And its working fine.  I suspect this can be closed now as the correct code is coming down now.

closed 2022-01-07 by mmozzano

Gadgetoid, 2022-05-11:
I've shipped a new displayotron library that depends upon `cap1xxx>=0.1.4` Should solve this!