pimoroni/displayotron #69: Does not install with pip and sudo

transilluminate, 2024-01-02:
Hi! I have a dot3k, and I'm re-installing on a Pi4 / Raspberry Pi OS Lite (64bit Debian Bookworm).

I needed to alter the script at `get.pimoroni.com/displayotron` to install. After each `sudo pip install` command I needed to add a flag to allow it to install to the system packages. See the diff:

```
822c822
<             if ! sudo -H $PIP2_BIN install --upgrade "$piplibname"; then
---
>             if ! sudo -H $PIP2_BIN install --upgrade "$piplibname" --break-system-packages; then
833c833
<             if ! sudo -H $PIP3_BIN install --upgrade "$piplibname"; then
---
>             if ! sudo -H $PIP3_BIN install --upgrade "$piplibname" --break-system-packages; then
890c890
<                     sudo -H $PIP2_BIN install "$moredep"
---
>                     sudo -H $PIP2_BIN install "$moredep" --break-system-packages
898c898
<                     sudo -H $PIP3_BIN install "$moredep"
---
>                     sudo -H $PIP3_BIN install "$moredep" --break-system-packages
907c907
<                 sudo -H $PIP2_BIN install "$pipdep"
---
>                 sudo -H $PIP2_BIN install "$pipdep" --break-system-packages
910c910
<                 sudo -H $PIP3_BIN install "$pipdep"
---
>                 sudo -H $PIP3_BIN install "$pipdep" --break-system-packages
```

Perhaps a more elegant way would be to use a venv, but the board needs root access anyway...! 😆

gazhay, 2024-08-14:
Even after this patch I still can't get it to install :-(

Gadgetoid, 2025-01-08:
It would be better to lean into using a virtualenv and install the packages directly with pip rather than using the installer. I wrote some possibly useful virtualenv stuff here: https://pimoroni.github.io/venv-python/

Requirements for root access should be a thing of the past, but the guide linked does cover running things from a virtual environment as root.